creepymonster/GlucoseDirect #3: readRaw() and writeRaw() not supported by the Libre 2

gui-dos, 2021-10-03:
The custom NFC commands `A3` and `A4` to read from any 64-bit address and to gain write access to the FRAM are not supported by the Libre 2.

The overhead introduced by the nested handlers at https://github.com/creepymonster/LibreDirectApp/blob/d1a94370ed8465234d967729fea277df121cc665/LibreDirectLibrary/Modules/Libre2/Libre2PairingService.swift#L95-L97 can be therefore avoided.

creepymonster, 2021-10-03:
Hi gui-dos, thanks for your message. I will look at your tip tomorrow and adjust and test the code. The customization sounds fine and would make the code more readable. I would like to make the code more readable again for the Libre2 in the near future and this would be a first step.

Many greetings

gui-dos, 2021-10-04:
I rewrote my [NFC.swift class](https://github.com/gui-dos/DiaBLE/blob/master/DiaBLE/NFC.swift) by using `async / await` and gathered the Libre 1 “hacks” in the separate [NFCTools](https://github.com/gui-dos/DiaBLE/blob/master/DiaBLE/NFCTools.swift). The code is still messy and I still have to solve all the warnings caused by the new *Actors* isolations...

May I ask you whether you already tried to support scanning the sensor to retrieve manually the last 8-hour data and to integrate the new Swift 5.5 `Tasks` in your Redux custom implementation?

creepymonster, 2021-10-04:
> Hi gui-dos, thanks for your message. I will look at your tip tomorrow and adjust and test the code. The customization sounds fine and would make the code more readable. I would like to make the code [...]

Hi gui-dos, I implemented your recommendation and it works wonderfully. Thanks a lot :)

creepymonster, 2021-10-04:
> I rewrote my [NFC.swift class](https://github.com/gui-dos/DiaBLE/blob/master/DiaBLE/NFC.swift) by using `async / await` and gathered the Libre 1 “hacks” in the separate [...]

I'm eagerly following your adjustments in the NFC class. I think async/await will improve the readability by a lot - but I will probably first change the Redux implementation to async/await before I can give myself to the NFC class.

I had already thought about the 8 hours of data, but I didn't tackle the issue to avoid the problem of duplicate data (with small differences, maybe not exactly the same data) being uploaded to Nightscout.

creepymonster, 2021-10-04:
> I rewrote my [NFC.swift class](https://github.com/gui-dos/DiaBLE/blob/master/DiaBLE/NFC.swift) by using `async / await` and gathered the Libre 1 “hacks” in the separate [...]

I must say that the 8-hour values have often bothered me in the past with Diabox. For example, when switching from xdripswift to Diabox, I had two curves over the last few hours in my Nightscout.

creepymonster, 2021-10-04:
I had also seen your hint at xdripswift where it was about rawValue == 0. I have logged that at least. I had fortunately exactly at the time the problem that I got only 0 values from my sensor (connected but no more values through the 0 filter).

WORK_VOLTAGE, SENSOR_SIGNAL_LOW

gui-dos, 2021-10-05:
I noticed that when @bubbledevteam ’s glucose.space replies with 0 values their dataQuality is 32768.

I think therefore that your redefinitions

https://github.com/creepymonster/LibreDirectApp/blob/44e3768acaf24bcf53c51cbd4ae420d692eff24d/LibreDirectLibrary/Modules/Libre2/Libre2Utility.swift#L318-L321

are not correct and that you should keep the LL definitions for the higher bits:

https://github.com/gui-dos/DiaBLE/blob/823e3452a6662b1713fb0adbf4473e3dc1440a49/DiaBLE/Glucose.swift#L33-L37

```
        static let THERMISTOR_OUT_OF_RANGE = DataQuality(rawValue: 1 << 11)
        static let TEMP_HIGH           = DataQuality(rawValue: 1 << 13)
        static let TEMP_LOW            = DataQuality(rawValue: 1 << 14)
        static let INVALID_DATA        = DataQuality(rawValue: 1 << 15)
```

But you are correctly using OptionSets for the DataQuality: I was wrong and copied & pasted your definitions :-)

creepymonster, 2021-10-05:
You are absolutely right - grrr, only the first lines in the eye had and the last lines carelessly numbered through.

INVALID_DATA = 32768 is not 1 << 12 but 1 << 15

creepymonster, 2021-10-16:
> I rewrote my [NFC.swift class](https://github.com/gui-dos/DiaBLE/blob/master/DiaBLE/NFC.swift) by using `async / await` and gathered the Libre 1 “hacks” in the separate [...]

I tried the NFC scan once with async/await. Actually looks quite good and works quite well in my tests.