creepymonster/GlucoseDirect #26: Schieber "Bewegung"

quotengrote, 2021-12-08:
Moin,

danke für die App, eine schöne Alternative zu xDrip.

In den Einstellungen habe ich ein interessantes Verhalten festgestellt.

Wenn man unter "Glukose-Einstellungen" einen der beiden Slider bewegt, verschiebt sich der andere mit, aber der Wert bleibt gleich.

Gruß
mg

creepymonster, 2021-12-08:
Hallo quotengrote,

danke für deine Nachricht. An dieser Stelle ist das ein "Feature" :) - wenn man den "Niedrig" Slider verschiebt, verschiebt sich das untere Ende der "Hoch" Skala (und anders herum auch). Es sieht so aus als würde sich der Wert des Hoch Sliders verändern, dabei wird nur der Bereich kleiner in dem der Hoch Slider bewegt werden kann.

Wenn ich also Niedrig von 80 auf 100 Schiebe lässt der Hoch Slider nur noch Werte zwischen 100 und 500 zu (vorher zwischen 80 und 500).

So ist es nicht mehr möglich den Hoch Slider auf 80 und den Niedrig auf 100 zu stellen.

Viele Grüße
Reimar

closed 2021-12-08 by creepymonster

quotengrote, 2021-12-08:
Aaah, klar, das klingt sinnig. Danke.

Planst du vllt die Möglichkeit auch eingenomme KH einzugeben mit einzubauen? :-)

creepymonster, 2021-12-08:
Das habe ich leider nicht geplant. Ich kann ja mal alles sammeln was die Funktion alles hergeben müsste damit sich daraus ein Mehrwert ergibt.

Im Moment versucht ich mit meiner begrenzten Zeit die App so übersichtlich und getestet wie möglich zu halten und wenig Bereiche zu schaffen die ich nicht auch selbst im Auge behalte weil ich sie nicht nutze.

quotengrote, 2021-12-08:
Falls ich irgendwie helfen kann, sag Bescheid, ein paar Ideen hätte ich, aber verstehe natürlich deine Begründung.

creepymonster, 2021-12-08:
Darauf komme ich gerne drauf zurück.

Ich werde mal schauen ob ich hier im Diskussionsbereich ein Sammelthema für Ideen erstellen kann.