creepymonster/GlucoseDirect #37: adding manual BG while waiting for sensor

mountrcg, 2021-12-09:
I was waiting for a virtual sensor connection and added a manual Glucose Reading in List. This initiated the graph on Overview with the sensor connection state output below
[image]

mountrcg, 2021-12-09:
So I guess the issue is the tint of the footer, which after the sensor is initialized looks fine again
[image]

creepymonster, 2021-12-09:
Thank you for the second screenshot. I don't know yet exactly why the navigation in the simulator (only in the simulator I could observe the problem) has no background color. But I haven't really been able to find out anything about it yet either.

closed 2021-12-09 by creepymonster

mountrcg, 2021-12-10:
Also on iphone not just simulator
[image]