creepymonster/GlucoseDirect #63: Libre US 14 days, Libre Pro/h

Bubblandevteam, 2021-12-15:
Requesting support for Libre US 14 days, Libre Pro/h with Bubble transmitter.

creepymonster, 2021-12-15:
I am sorry, I am afraid I will not be able to provide this support. I own a Bubble Transmitter (with which I could develop and test the connection), but I have no access to Libre US 14, Libre Pro/h sensors to deal with.

Does the Bubble Transmitter (depending on firmware version) support decryption of both sensors (via command [0x08, 0x01, 0x00, 0x00, 0x2B] - similar to Libre 2)?

bubbledevteam, 2021-12-16:
> I am sorry, I am afraid I will not be able to provide this support. I own a Bubble Transmitter (with which I could develop and test the connection), but I have no access to Libre US 14, Libre Pro/h [...]

yes,you can get decrypted data with 0x08 command for Libre 2 EU and Libre US 14 ,but I don't recommend use this with LibrePro for I don't finish the glucose Algorithm for this type so far

creepymonster, 2021-12-16:
Thanks a lot, I will have a look and then implement it.

closed 2021-12-16 by creepymonster