creepymonster/GlucoseDirect #71: nightscout upload doesn't work

midnightcoup, 2021-12-16:
Hello,

Thank you for this app. I noticed that nightscout upload isn't working even if I added the url and API secret. Is this expected?

Regards.

creepymonster, 2021-12-16:
Thanks for the info. Both must be entered - Nightscout URL and API_Secret.

The URL for example in the form:
https://my-nightscout.herokuapp.com

If that doesn't help I would be happy if you could send me the logfile under "Settings" at the bottom. To reimarmetzen@gmail.com

Many greetings

berpj, 2021-12-16:
> The URL for example in the form: https://my-nightscout.herokuapp.com
Thanks, at first I didn't enter "https://" and it wasn't working for me neither.

It says "Host" in the app: I think "URL" would be a better description, and maybe a placeholder could be useful.

Thank you for this app.

creepymonster, 2021-12-16:
That is a good idea. I will include a placeholder and pay attention to the protocol - renaming is also a good hint. I will implement the suggestions for improvement and validate the inputs.

Many greetings

closed 2021-12-16 by creepymonster

creepymonster, 2021-12-17:
I implemented your suggestion in the next update.

midnightcoup, 2021-12-20:
Ah, yes. It was https:// missing. Thank you.