creepymonster/GlucoseDirect #74: "the sensor sends faulty values" after scanning with Librelink

midnightcoup, 2021-12-16:
Title explains it. One reading after scanning with the LibreLink app always says that the reading is faulty.

creepymonster, 2021-12-16:
Can you describe the problem in more detail? After scanning with LibreLink, GD shows the exclamation mark in the list of values? This can happen occasionally and should go away.

closed 2021-12-17 by creepymonster

midnightcoup, 2021-12-20:
I know that it happens occasionally on the Libre where even the Libre app doesn't give readings under certain conditions, particularly when it is too hot (in a sauna etc). 
But in this case, it consistently happens for exactly 1 readings in the GlucoseDirect app after every LibreLink NFC read. So I felt like this was systematic bug in the app rather than the sensor actually sending faulty values.

creepymonster, 2021-12-20:
Thank you, now I understand how the problem occurs with you.

I tried it a few times on my computer and could not reproduce the problem.

In the case of faulty values, the sensor returns a 0 as the glucose value and in another field the cause of the error. Therefore, it should not be a wrongly recognized value.

I will keep an eye on the problem, maybe I can reproduce it after all.

Many greetings