creepymonster/GlucoseDirect #80: Libre algorithm

dugamarian, 2021-12-18:
It's a great app and very useful, but I have some suggestions to make if I'm allowed. First, and most importantly is the libre algorithm which in my opinion is reading to low (with about 6-7%). It would be great if you could raise the readings with about 6% at least.   
Secondly but not so important are the notifications which should be more basic. Besides that congratulations for your work!

creepymonster, 2021-12-18:
Thanks a lot for your feedback.

I cannot increase the values for all by about 6%. For some this might work great, but for others it might not. You can achieve this effect for you by calibrating your sensor. The best time is during a phase of stable blood glucose values. Measure with a blood glucose meter and store the measured value in the calibration area (as shown on the screenshot).

What do you mean by "more basic" - can you give me an example of this. Actually, I already found the notifications to be very basic :D

[image]

dugamarian, 2021-12-18:
I mean more besic like just glucose reading, unit and Delta. Now it's showing also the description. Anyway, it's not a big deal. Thanks for your work!

creepymonster, 2021-12-18:
Thanks, it's good to know that. I was afraid that I would be a bit too brief in the notification

closed 2021-12-18 by creepymonster