creepymonster/GlucoseDirect #83: Nightscout upload improvements

blackmondaytom, 2021-12-18:
First big thanks for the app. Really love this app. Simple, and supports everything I need (show current glucose value, simple calibration possibility, and upload data to nightscout).

For the nightscout upload, I've some sort of feature request.
It seems that the glucose value is uploaded every 5 minutes (00:00 - 00:05 - 00:10 - ... 23:50 - 23:55).

Whenever such an upload should be done, but there's some issue with the glucose value (e.g. sensor is too far away from the phone and no bluetooth connection is available) or with the phone itself (e.g. no cell phone signal and no WLAN, or airplane mode is turned on), the value is missing in nightscout (and in the Diagram area of the app).
The feature request would be to detect such issues, and provide (e.g. upload to nightscout) the data at a later point when the issue is resolved.

Thinking about possible solution resulted in the following ;-) ... This may be 2 different scenarios:
a) no bluetooth connection to sensor -> in this case you don't have the data. But maybe there some way to get "historical data" from the sensor (last 8h), and upload the missing pieces.

b) no signal/airplane mode -> here the data should be available in the app. Delayed uploading the data, when either WLAN or cell phone signal is restored.

I don't know about the technical challenges for this 2 scenarios, and how difficult it is to find out if upload to nightscout was not possible or failed.

What's your opinion on this, do you think this may be something that could be supported?

Thanks for the great app!

creepymonster, 2021-12-18:
Thank you, those are two very good suggestions. I would tackle the suggestions separately. Both are good to implement and will be written by me as next items on the todo.

I think first the filling of the gaps when the sensor was not connected, and then the subsequent upload of missing data to Nightscout.

blackmondaytom, 2021-12-18:
Yes, implementing them separately absolutely makes sense.

If you need some "tester", I would be available :-)

creepymonster, 2021-12-19:
I'll be happy to come back to that ☺️.

If you compile the app yourself, the main branch already contains the function that on reconnection all missing values (sent via BLE) are filled in the gaps.

blackmondaytom, 2021-12-21:
Hab die App einmal kompiliert, ausprobiert und dir ein Mail mit den Beobachtungen geschrieben.

paulus81, 2022-02-26:
Hi there creepymonster,

First of all thanks for developing the "Glucose Direct App" (it´s a blessing!).

Probably I am missing something but I am trying to use the "Upload to Nightscout" feature and it is not working.

Any advice on what could be wrong or what can we check to find the issue and solve it.

Thanks!

vdcunha, 2022-03-06:
Although my Nightscout url begins with http:// in the app I had to write https://
Congrats to CreepyMonster and all the team for the nice work!