creepymonster/GlucoseDirect #92: HealthKit integration

asiergmorato, 2021-12-20:
Please, consider adding a HealthKit integration for saving glucose values on Apple Health.

creepymonster, 2021-12-20:
Thank you very much for the feedback. The Apple Health connection is something I definitely want to address. In the current phase, I'm working on stabilizing the app and implementing smaller features and fine-tuning existing ones. After that, Apple Health is very high on my wish list.

Many greetings

asiergmorato, 2021-12-21:
Thank you for making the app! 🤗

asiergmorato, 2022-02-05:
Just shaw that it is available on TestFlight. There is a small bug: The app does not respect the reading interval an saves data to Health every minute even if the app is configured to read every 5 or 15.

closed 2022-12-11 by creepymonster