creepymonster/GlucoseDirect #109: Nightscout integration

BBW20, 2021-12-24:
I am unable to use nightscout alongside this app because the api secret password box will only take 10 characters not 12 on my phone.

creepymonster, 2021-12-24:
Hmm, my API Secret is 25 characters long. I can imagine that only 12 characters (regardless of the actual length) are displayed.

You can send me a logfile once.

It is important that you specify the Nightscout URL with scheme (https://).

Do you use a Nightscout token instead of the API secret?

Many greetings
Reimar

BBW20, 2021-12-24:
Do you need me to send my nightscout thing, I my password is the same for both and it doesn’t fit.

creepymonster, 2021-12-24:
For now, it would be enough for me if you send me the logfile under Settings below to reimarmetzen@gmail.com. I would like to avoid getting access to your Nightscout.

BBW20, 2021-12-24:
Here’s the log from the app

Brooklyn

creepymonster, 2021-12-24:
Unfortunately, I do not know what the text means.

Look under settings, then click on the highlighted item and there select mail and send to me.

Many greetings @
[image]

creepymonster, 2021-12-24:
[image]

creepymonster, 2021-12-24:
[image]

creepymonster, 2021-12-24:
Use due TestFlight or self compiled me version. In the self-compiled version I could build in the authentication via token relatively fast. At Apple, the release over the Christmas days unfortunately takes a little longer.

BBW20, 2021-12-24:
Hey I’m not sure why it just sends the text.
I’m using the TestFlight link if that’s helpful.

Brooklyn Whyte

creepymonster, 2021-12-24:
Did you create the secret via the browser in the Nighrscout administration, or is that the setting at heroku? Does the role have the correct roles at Nightscout?

BBW20, 2021-12-24:
Heroku everything is correct everywhere else I can log into nightscout I just can’t connect it to glucose direct

Brooklyn Whyte

creepymonster, 2021-12-25:
Ah, I think I understand now what the text below means. You are replying to an email from GitHub to reply to me.

Can you go into the app once at Glucose Direct, settings and at the very bottom click on send bug report. Then select Mail and enter reimarmetzen@gmail.com as the recipient.

creepymonster, 2021-12-25:
I can close the ticket as the problem has resolved itself. 

Merry Christmas

closed 2021-12-25 by creepymonster