creepymonster/GlucoseDirect #120: Libre Link Upload&Dowbload integration

yoshimo, 2021-12-26:
For relatives and doctors that are not as technically capable and willing to use opensource software, it would be great if you could interface fitectly with the Abbott database.
Being a Libre Linkup partner to watch over your relatives glucose levels or send your own measurements to your linkup partners like wife, husband, boyfriend or doctor would be great.

creepymonster, 2021-12-29:
I don't want to include this feature, at least not as long as there is no semi-official way. Otherwise it is a guarantee for big trouble (possibly banned user accounts or ban from TestFlight). Even if I find the possibility very useful, I can also understand the anger of the manufacturer.

Many greetings

closed 2021-12-29 by creepymonster

yoshimo, 2021-12-29:
What do you mean by "semi-official" in this context?

creepymonster, 2021-12-29:
I thought I had seen something about a possibility of CSV or Excel import. But maybe it was only an export from LibreView.

I could also imagine that a Nightscout plugin could close the gap quite well.