creepymonster/GlucoseDirect #121: Make timescale longer in default graph view/ make timerange selectable

midnightcoup, 2021-12-26:
The current time range in the graph is not very long (around 1h) and is not really enough to see any meaningful trends in blood glucose. The graph is often just an almost straight line. It is not really possible to notice any trends or have an overview of the day. 

Ideally, the default view should be set to 6h and there should be a selectable timescale of 3h, 6h, 12h, and 24h.

creepymonster, 2021-12-27:
I like the suggestion very much, I think this will be one of the first new features I will implement. The graph is really very detailed. I think this will be one of the first functions I will implement.

creepymonster, 2021-12-29:
I am in the process of developing the feature and am currently testing it in my app. It is already included in the GitHub code in a simple version. After testing and fine-tuning, the function will also be included in the TestFlight version.

closed 2021-12-29 by creepymonster