creepymonster/GlucoseDirect #133: erratic trend arrows

midnightcoup, 2021-12-28:
It seems that the trend arrow is based on the difference compared to only the last reading. This is not the best implementation because for example, if I leave the house, the change in temperature creates a small but sharp variation in in BG measurement and this frequently shows up as ↑↑ or ↓↓. The measurement soon stabilizes, but there is this transient time where the arrows are shown erratically.

This can be mitigated by using last 10 mins or so of readings as moving mean average to calculate the trend.

creepymonster, 2021-12-29:
A trend over several minutes makes sense in any case. So far, I had favored the option because it was very easy to follow. It will definitely be on my list.

Fielder84, 2023-03-03:
@creepymonster Thank you so much for your work on this app! Any idea on solving the erratic trending arrows? I agree if it was an average over a longer period it would it would be great. Currently I can not trust it, so it’s the only thing I miss from the official app. Thanks again!