creepymonster/GlucoseDirect #233: A test “companion” mode for Libre 2 / LibreLink.

gui-dos, 2022-06-15:
If you just activate the notifications on the Libre 2 read characteristic without sending the unlock payload and then launch LibreLink, you can receive the data regularly while keeping both apps running side-by-side, even after closing LibreLink.

https://github.com/creepymonster/GlucoseDirect/blob/cdb8ee8bbabde9887dd9fa06eec3b86c78d65021/App/Modules/SensorConnector/Libre2Connection/Libre2Connection.swift#L127

This is technically possible on iOS with any app since in CoreBluetooth the connections are reference-counted and a device has no clue that the same characteristics’ values will be notified to all the multiple apps that subscribed to them.

It could be a way to test the app without activating BLE on your own since LibreLink can keep on reconnecting successfully to the sensor and turning off the red exclamation mark regarding the alarms unavailability in a couple of minutes.

I implemented it very easily in the “Test” mode of https://github.com/gui-dos/DiaBLE so I think that porting it will be quite feasiBLE :-)

creepymonster, 2022-06-15:
Hi gui-dos,

do you connect to the already connected sensor with .retrieveConnectedPeripherals - or do you get already connected sensors with .retrievePeripherals? 

Many greetings
Reimar

gui-dos, 2022-06-15:
I am just letting the Libre 2 disconnect since it is expecting a BLE login and I try to connect again to it. In the 2-minutes meantime I launch LibreLink (but it worked also with DiaBox i.e. when it was available on TestFlight) and the BLE data are notified automatically. To decrypt them you need of course the sensor UID (which the sensor includes in the advertised manufacturer data anyway) and a preliminary NFC scan is useful to get the factory calibration parameters and the last 8-hour historic data, of course.

It is the same technique I am using to sniff the Libre 3 incoming BLE traffic.

I had to rewrite my naive Log class that used just a single @published String: with a LazyVStack and multiple Text lines luckily I can analyze now flawlessly the much larger amounts of transmitted data without freezing B-)

creepymonster, 2022-06-16:
Hello gui-dos,

I tried it yesterday - I had tried reconnecting an already connected sensor in the past. But there I had a thinking error at the point with the .setNotifyValue

It works also very well - yesterday I was still facing the problem how to recognize if an unlock is necessary (because the connection to the sensor is established by GD) or if the unlock can be omitted because it is already a (from another app) connected sensor.

At the moment it works quite well by checking after .retrievePeripherals with .retrieveConnectedPeripherals if the sensor is already connected or not.

I can't judge yet if the whole thing runs stable - the app that connects the sensor has to be open all the time. If e.g. LibreLink is closed accidentally and the connection to the sensor is lost, GD should not be able to connect to the sensor anymore, because the unlock info is no longer correct.

I will experiment a bit with it and then check in my adjustments soon.

gui-dos, 2022-06-16:
I wanted to mean not to connect to a Libre 2 that other apps are already using or that Glucose Direct already paired.

My idea is that, before activating the BLE connection on your own and breaking the other apps since their unlock codes and counters will not be recognised anymore by the sensor, it would be nice to be able to intercept their data telling the final users that there is no risk to "break" the Libre 2 connections. The user can just uninstall and reinstall Libre Link to get the alarms restored but he could run also Glucose Direct in parallel with it or with Shuggah, or zDrip, or any other app by other bad guys who aren't warning the users that they are going to deactivate the alarms of other apps ...

This is a quick movie I posted also in https://github.com/JohanDegraeve/xdripswift/issues/362 which shows how I am launching LibreLink after 1.5 minutes the sensor has disconnected: initially LibreLink displays a red exclamation mark but you can see that I can receive the data at the 120th second, upload them to Nightscout and, after switching back to LibreLink, the exclamation point disappears since the alarms keep on being played regularly:

https://user-images.githubusercontent.com/7220550/174006255-f4f9f9db-2744-4c13-9297-3e05ae8f0dcc.mov

creepymonster, 2022-06-16:
But that always requires that LibreLink (or someone else) connects the sensor and does the unlock? If there is not this app that connects the sensor and "unlocks" there is never a connection to stand?

gui-dos, 2022-06-16:
That’s exactly why I am calling this warkaround a “test companion” mode. I would be better name it perhaps an “opportunist parasite” who lets the others do the dirty work of unlocking and keeping the connections :)

creepymonster, 2022-06-16:
I'm afraid that there might be some problems if LibreLink is not always running. When LibreLink is closed, the Bluetooth connection remains through the reference-count of iOS. But if the connection is lost for some reason, no new connection to the sensor can be established without the LibreLink app.

I could imagine a "transmitter" LibreLink creation - there is then in the NFC scan then no .enableStreaming made, and the connection always takes place without unlock.

gui-dos, 2022-06-16:
Why a “LibreLink Trasnmitter” class? The workaround is valid for any app.

Tonight LL and DiaBLE got along all the time so this morning I decided to enable the BLE connection by using Glucose Direct since I spent the first days solving the bugs I haven’t been testing for months (it is my last spare Libre 2 from the last year since the Libre 3 will be soon on sale here in July officially).

I lost the LibreLink alarms but, as you can hear in the quick movie, I got two, both from Direct Glucose and DiaBLE running and playing sounds along side-by-side! ;-) 

https://user-images.githubusercontent.com/7220550/174039142-d05d3f10-466b-4839-b54d-3b6212175af5.mp4

gui-dos, 2022-06-18:
Two bad hypos the last night but luckily it was a nice morning when I tested your commit https://github.com/creepymonster/GlucoseDirect/commit/0ddf163674a73a39f5faa2195bc189c8bc60c70a.

You grasped my idea and implemented it brilliantly: testing Glucose Direct and LibreLink together without those nasty exclamation marks and the need to reinstall LL and loosing the local database data :-)

[image]

creepymonster, 2022-06-18:
Thanks gui-dos, 

it was fun to try out your idea. The exchange with you is very important to me and always shows me completely new aspects.

I had tried your suggestion before, but still had a bug then - your confirmation made me look into it again.

Many greetings
Reimar

closed 2022-06-20 by gui-dos