creepymonster/GlucoseDirect #293: Compilation Error

avouspierre, 2022-07-21:
Xcode v13.4.1 
Main branch 

I try to compile the code with my Iphone XR IOS 15.5. I fail to compile with lot of errors, in particular : 

`Glucose.swift:15:30: Protocol 'Glucose' can only be used as a generic constraint because it has Self or associated type requirements`
`TransmitterWidget.swift:74:28: Variable binding in a condition requires an initializer`
`GlucoseWidget.swift:144:59: Type 'WidgetFamily' has no member 'accessoryRectangular'`

I just updated the GlucoseDirect.xconfig for the development team. 

[image]

Any idea ?

creepymonster, 2022-07-21:
Hello,

I already have some changes for iOS 16 in preparation. To build the app therefore already the last Xcode 14 beta is needed (can also be installed in parallel), but the app still runs on iOS 15.

Many greetings
Reimar

avouspierre, 2022-07-22:
Thank a lot ! Indeed  with Xcode 14., the compilation was succeeded.

But, I have a error in the debug mode : 
Row.swift:361

`Thread 7: Fatal error: 'try!' expression unexpectedly raised an error: could not decode Date from database value NULL - column: "fromTimestamp", column index: 1, row: [readings:0 fromTimestamp:NULL toTimestamp:NULL gmi:NULL avg:NULL tbr:NULL tar:NULL days:NULL variance:NULL]`

creepymonster, 2022-07-23:
Thank you for pointing this out, this is a bug that neither I nor the testers noticed as we all have data in our databases already.

I have just created and committed a fix for it

Many greetings
Reimar

closed 2022-07-23 by avouspierre