creepymonster/GlucoseDirect #294: GlucoseDirect with Loop - Dev

avouspierre, 2022-07-23:
I try to use GlucoseDirect with Loop - Dev with the glucoseDirect client plugin. 

I removed the Carthage script in the plugin and the Glucose Direct appears in the list of the CGM. But, unable to go after... 

`2022-07-23 10:19:23.709730+0200 Loop[9671:432515] [StatusTableViewController] Failure to setup CGM manager with identifier 'GlucoseDirectClient': UnknownCGMManagerIdentifierError()`

Do you know in GD works with Loop -Dev (Loop 3) ? 

Thanks,

avouspierre, 2022-07-23:
I found the solution. With dev, we didn't require Carthage but you need to embedded the Glucose Direct framework inside the plugin. Sincerely,

[image]

creepymonster, 2022-07-23:
Thank you, I will adjust the client accordingly.

I have not updated Loop yet and still use the dev version with Carthage.

Many greetings
Reimar