creepymonster/GlucoseDirect #322: Crash

egpall, 2022-07-28:
Ayer instalé iOS 16 beta 4 y GlucoseDirect crash

egpall, 2022-07-28:
He vuelto a la versión anterior de GlucoseDirect y no crash

creepymonster, 2022-07-28:
Which version did you have installed? I had published the 1.0.0 (102) via TestFlight yesterday, which works for me with the iOS 16 beta.

This morning I had already received feedback that the latest TestFlight version no longer crashes after a reboot of the phone under iOS 16.

Many greetings
Reimar

egpall, 2022-07-28:
Después de instalar la 1.0.0 hacía crash, apagué el teléfono y seguía igual, voy ahora a meterla de nuevo y a resetear el teléfono. Muchas gracias.

egpall, 2022-07-28:
He instalado la 1.0.0 y reseteado el iPhone con iOS 16 beta 4 y sigue crash, no puedo enviar la BD porque no hace crash

creepymonster, 2022-07-28:
Unfortunately, I have no clue yet what has changed that can lead to problems in places.

I will try to find out more from the crash reports.

Many greetings
Reimar

egpall, 2022-07-28:
Gracias. Magnífica App.

egpall, 2022-07-28:
Voy a cambiar a idioma inglés, a ver si se soluciona

egpall, 2022-07-28:
Sigue igual con idioma inglés

egpall, 2022-07-28:
He reiniciado también

creepymonster, 2022-07-28:
I'm afraid I'm going to have to fumble my way a bit. I have created a new build (105).

Should the build start for you it would be great if you could send me your DB under settings.

Many greetings
Reimar

egpall, 2022-07-28:
Cuando me salga en TestFlight la instalo y te envío la BD, todavía me sale la 100.

creepymonster, 2022-07-28:
Which is the latest version you can install via TestFlight - do you already see versions 102, or later?

egpall, 2022-07-28:
Acabo de instalar la 102 y ya funciona bien!!!! No crash!!!!! Gracias

creepymonster, 2022-07-28:
That's good ☺️, then I had a thinking error at noon today 😁.

I thought you are in the iOS 16 test group and already have access to build 102, but you are registered normally via TestFlight and could not yet get the latest update.

Until now, it was always necessary to rebuild the app once with a new beta.

closed 2022-07-28 by creepymonster

egpall, 2022-07-28:
Sí, soy un usuario normal a través de TestFlight