creepymonster/GlucoseDirect #337: Calibration

egpall, 2022-07-31:
The first thing to say is that I think the application is wonderful. I wanted to make a query and yes, it is necessary to record some calibration and if so, it would be convenient, several measurements at different times of the day with test strips and write them down or what would be the best method. Mucha I wanted to make a query and yes, it is necessary to register some calibration and if so, it would be convenient, several measurements at different times of the day with test strips and write them down or what would be the best method. Thank you very much for everything. Thank you for everything.

egpall, 2022-08-06:
Can anyone give me any advice on this? Thank you

creepymonster, 2022-08-06:
Hello,

Calibrations are not really necessary. This is only the possibility to "optimize" the values a bit more.

I would advise most people not to calibrate, because incorrect calibrations can also be a great danger.

Are you looking for tips on when and how best to calibrate?

Many greetings
Reimar

egpall, 2022-08-06:
Thank you. That's what I wanted to know and confirm.

closed 2022-08-06 by creepymonster