creepymonster/GlucoseDirect #352: Suggestion integration Calendar

egpall, 2022-08-06:
The integration with Calendar seems magic to me, it is wonderful, even in CarPlay it gives me the current values, but I have a suggestion, the highest refresh time is 15 minutes and in Calendar the values for 10 minutes appear so there are 5 minutes in which there is no data, if the times coincided there would be no times in Calendar without values. Could it be done? Thank you

creepymonster, 2022-08-06:
I do not know yet if I understand the problem correctly.

The app itself updates the calendar every minute - every minute the calendar entry is updated. So you can always see the latest values on the Apple Watch, for example.

Is there a limit in Car? Or where do the 10/15 minutes come from? And how can I imagine that the gaps arise?

egpall, 2022-08-06:
I have said about CarPlay only as a compliment, I add 2 photos where you can see that my calendar assigns 10 minutes to each event and as I refresh the value in GlucoseDirect every 15 minutes there are 5 minutes without values. Why do you create the 10-minute calendar events for me? Have you programmed that time or is it my iPhone? Thank you again

[image]
[image]

creepymonster, 2022-08-06:
Ah, I see - the 10 minutes displayed there are just the length of the calendar entry. This does not create any gaps.

An entry in the calendar must have a start and end time - I have only selected 10 minutes there. The actual length of the entry has no meaning.

Even if you set the interval to 1 minute, the calendar entries have a length of 10 minutes.

I will adjust it so that the length of the entry corresponds to the interval of the sensor.

egpall, 2022-08-06:
Perfect!!!!!! Thank you again

creepymonster, 2022-08-07:
I have made the adjustment - with the next update the length of the calendar entry corresponds to the sensor interval.

Many greetings
Reimar

closed 2022-08-07 by creepymonster

egpall, 2022-08-07:
Perfect. Thank you very much.

egpall, 2022-08-25:
Sigo teniendo tiempos sin valores en Calendario, creo que se solucionaría si para el intervalo de 1’ en Calendario fueran 2’, para 5’ fueran 8’ y para 15’ por ejemplo 20’. Es sólo una sugerencia.