creepymonster/GlucoseDirect #354: Crash in iOS 16 beta 6

egpall, 2022-08-16:
He vuelto a la 102, las posteriores hacen crash

creepymonster, 2022-08-21:
Sorry for my late reply, I was on vacation and traveling with the RV for the last 14 days.

Unfortunately, there are issues with the ActivityKit (which is responsible for displaying the real-time glucose updates on the lockscreen) on iOS 16 Beta 6. I have disabled ActivityKit for now until a workaround is known or the issue is fixed with a future beta.

closed 2022-08-21 by creepymonster

egpall, 2022-08-21:
Thank you and I hope the vacation has been great!!!!

creepymonster, 2022-08-21:
Oh yes, thank you very much - we had a lot of fun - my first trip to Denmark was very nice :)

The version 1.0.0 (116) fixes the problem for now.

I hope that the display of current values via the ActivityKit will soon be possible again. Until then, I had (at least for now) only the possibility to disable the function :(

egpall, 2022-08-21:
I've already updated it and it's perfect!!!!! Thanks a lot

egpall, 2022-08-21:
I love the data, I export the database, extract Date, Time and Value, correct the time offset and paste everything into a spreadsheet. I am attaching capture.

egpall, 2022-08-21:
I love the data, I export the database, extract Date, Time and Value, correct the time offset and paste everything into a spreadsheet. I am attaching capture.
[image]

egpall, 2022-08-21:
[image]

creepymonster, 2022-08-24:
I find the idea cool - I have to look at that for evaluations. So far I had used the SQLite DB mainly to quickly make a backup of the data to be able to test without data.

egpall, 2022-08-24:
I have no idea about programming, if I know what I want and what I like. With a few clicks I get an impressive amount of data. Your work is impressive.

egpall, 2022-08-24:
A question, for the calculation of HbA1C, what formula do you use? If you can give it, of course. Thank you for everything.

creepymonster, 2022-08-24:
I think the work is great - it doesn't even need a program to do it - I think the spreadsheet is just the ticket <3

I got the formula for the GMI from https://www.jaeb.org/gmi/

GMI(%) = 3.31 + 0.02392 x [mean glucose in mg/dL]
GMI(mmol/mol) = 12.71 + 4.70587 x [mean glucose in mmol/L]

egpall, 2022-08-24:
Thank you very much!!!!! I guess for a period of time having the average glucose you have the GMI.

creepymonster, 2022-08-24:
I also find the mean value together with the standard deviation very good. Both values give a quick overview of where you stand.

The standard deviation provides information about how much the values fluctuate around the mean.

Unfortunately, the GMI does not take such fluctuations into account. The two values 40 and 240 result in a good GMI - but via mean value and standard deviation you can see that the mean value is not so good.

creepymonster, 2022-08-24:
The lower the standard deviation, the less fluctuation in blood glucose levels - this is a value where you can easily say "the lower the better".

egpall, 2022-08-25:
Thank you for all your attention

w-aurelien, 2022-10-02:
Hello

I'm trying to compile myself, but I have too ActivityKit problem.

[image]

I use Xcode 14.0.1 and iOs 16.0.2.

Any solutions ?

creepymonster, 2022-10-02:
There were several Xcode versions with 16.0 where the ActivityKit (for real-time glucose widgets on the lockscreen) was not available. The 14.0.1 release unfortunately does not include the ActivityKit.

For example, you can use the Xcode 14.1 beta 3 release to compile the app for iOS 15 and 16.0 as well. With the beta of 16.1 you can also test the realtime lockscreen widgets.

[image]