creepymonster/GlucoseDirect #423: Kennzeichensymbol verschwindet

FriedelXT, 2022-11-15:
Hallo,

das kennzeichensymbol wird nicht zuverlässig angezeigt.
Mal ist es da, mal nicht.

aktuelle GlucoseDirect version

Schöne Grüsse
Friedhelm Walldorf

creepymonster, 2022-11-15:
Hallo,

um die Anzeige des Glukosewertes auf dem App Symbol auf für mmol Werte zu ermöglichen, gibt es nur die Möglichkeit iOS Benachrichtigungen zu nutzen.

Auf diesem Weg können mmol Werte angezeigt werden.

Wenn man jetzt aber zum Beispiel die normalen Glukose Benachrichtigungen ausstellt, erscheint der Glukosewert auf dem App Symbol nur noch wenn man zu hoch ist.

Hast du eventuell eine der folgenden Benachrichtigungen deaktiviert?

[image]

Viele Grüße 
Reimar

FriedelXT, 2022-11-15:
Hi Reimar,

genau das wars. Da bin ich Depp gar nicht drauf gekommen, weil ich an Alarm gedacht habe.
Danke für den Hinweis.

Schöne Grüsse
Friedhelm

creepymonster, 2022-11-15:
Leider ist Apple da nicht sehr konsequent - mit der einen Methode sind nur ganze Zahlen möglich, mit der anderen sowohl als auch.

Ich wollte das GD sich bei mmol und mg/dl Werten gleich verhält.

Da es wichtig war die Benachrichtigungen für normale Werte auszustellen (es gibt SmartWatches die ansonsten bei jedem Wert vibrieren) gab es nur die Möglichkeit in diesem Fall die Werte von dem Icon zu löschen. Ein veralteter Wert wäre auch nicht hilfreich.

So erschien es mir als der Beste Kompromiss 

Viele Grüße 
Reimar

closed 2022-11-15 by creepymonster