creepymonster/GlucoseDirect #424: Initializing Libre 2

apkuhar, 2022-11-19:
Is there a way to initialize a new Libre 2 EU sensor with this app? Or does it always have to be initialized with the official libre app?

gui-dos, 2022-11-19:
The `A1` “subcommand” to activate a Libre 2 sensor is `1B`.

While the `A0` NFC custom command to activate a Libre 1 requires to append an UInt32 password valid for all the sensors, the 4-bytes to append to the Libre 2 activation command are encoded taking into account the sensor NFC unique identifier according the algorithm I am outlining in https://github.com/gui-dos/DiaBLE/blob/62c78f6c110eb4ec19db1ca8b4075428b273daa3/DiaBLE%20Playground.swiftpm/NFC.swift#L111-L122

creepymonster, 2022-11-19:
I myself have always started the sensor with the LibreLink app and then connected it to GD.

After I had in DiaBox already funny results when starting (sensor 1970 started or so), I have not implemented the process itself and also never tried again to start the sensor with another app than LibreLink.

I thought, then at least the sensor is also recorded there as started (last serial numbers are displayed there). If in case of complaint is ever asked about it.

Many greetings
Reimar

closed 2022-11-19 by creepymonster