creepymonster/GlucoseDirect #427: "0 hours left" notification when sensor has 24 hours remaining.

t1mlopez, 2022-11-22:
Using Libre 3 (US) with LIbreLinkUp configured, when I hit the 24 hour left mark on my sensor, I received a notification warning me there were "0 hours" left. Unfortunately, I selected the alert so it was removed from my notification center and I couldn't get a screenshot. 

[Logs.txt](https://github.com/creepymonster/GlucoseDirect/files/10071108/Logs.txt)

creepymonster, 2022-11-22:
Thanks for the tip, I will take a closer look at the problem.

creepymonster, 2022-12-11:
There was an error in the display of notifications, which led to the fact that (with less than 24 hours) the remaining hours and minutes were not displayed. I am currently testing a fix, which I will publish with the next release.

closed 2022-12-11 by creepymonster