creepymonster/GlucoseDirect #442: No Data - State Disconnected

skidding, 2022-12-13:
Hello,

First of all thanks for this wonderful app!

Unfortunately I can't seem to get past the No Data phase. Scanned the sensor successfully and entered the LibreLinkUp credentials, too. I assume they're correct because I see a specific error when I purposefully enter wrong credentials.

Is there a reason why the warmup phase should take long? It's been almost 1h since I first scanned the sensor. As a comparison, I logged into LibreLinkUp right before this for the first time and data showed up immediately there.

This is what the app looks like:
[image]

- Log file with the email and name removed: [AllLogs.log](https://github.com/creepymonster/GlucoseDirect/files/10222740/AllLogs.log)
- iPhone 14 Pro
- iOS Version 16.1.2

Let me know if there's anything else I can provide or if you have any ideas. Thanks!

creepymonster, 2022-12-15:
Hello, 

Thank you for your message. I would assume that the connection between Libre 3 and LibreLinkUp is not yet complete.

It is best to install the apps Libre 3 and LibreLinkUp. In the Libre 3 app you must then enter a LibreView account under connected apps > LibreLinkUp. You have to use the same account in LibreLinkUp (and later GD).

It is very important to complete the setup in Libre 3 and LibreLinkUp (in the LinreLinkUp app you have to accept an invitation from the Libre 3 app. No data will be transferred before that).

Here is a video showing the whole process. Only when data appears in the LibreLinkUp app can GD receive any.

https://youtube.com/shorts/Ljj-eQ4Jl30

Many greetings 
Reimar

skidding, 2022-12-15:
Hi @creepymonster,

Thanks for the reply. I made some progress since I opened this issue.

I was hoping to use Glucose Direct as a LibreLinkUp alternative to monitor another person who uses Libre. I am not a Libre user and the report above is from my phone. Afterwards I managed to install Glucose Direct on phone of the person that uses Libre and it works – which is nice because we get extra functionality with this app.

Does Glucose Direct only work on the same phone with Libre 3 and/or LibreLinkUp view installed? If so you can close this issue. But I would appreciate if you could explain why. As I said I'm not a Libre user but I would like to be able to follow someone else and I figured since Glucose Direct connects over the LibreLinkUp API (which I have no knowledge of at the moment but would be interested to learn more) it should work from any phone.

Thanks again for you time!