creepymonster/GlucoseDirect #475: bluetooth is not working properly

jacun222, 2023-01-09:
Hi,
I Just installed glucosedirect version 140, previousy used xdripforios.
Comparing both glucosedirect stops working when I am 1,5 - 2 meters from my phone. Reconnecting take about 4-6 minutes.
Do u have plans to improve it?
Thanks for all you work, hope that app will be better regards to connectivity 
Jacun

philippkanter, 2023-01-11:
@jacun222 Even though I'm not the author of this app, I might be able to answer: Afaik glucosedirect dos not connect to the actual sensor via bluetooth to get the glucose readings. It uses the LibreLinkUp connection to get the values from the cloud. Hence the bluetooth range doesn't matter. I hope it makes sense and also is the correct reply. ;)

cc @creepymonster

jacun222, 2023-01-11:
Hi,
Are u sure? Even without installed LibreLink glucosedirect is still
working, as I read app connect directly to libre via Bluetooth
Lots of ppl have same problems (from Facebook groups are read it)
Jakub

philippkanter, 2023-01-11:
@jacun222 Sorry, I just realised, that glucosedirect works different for FreestyleLibre v2 compared to v3. FreestyleLibre v2 does uses bluetooth it seems.

jacun222, 2023-01-11:
No worries!
Still, I found app great! But smth is not as should be regards to
connectivity
Jakub