creepymonster/GlucoseDirect #486: Ascertain age > 0 before setting state to .starting

gui-dos, 2023-01-12:
If you try to scan an expired Libre 3 the line  https://github.com/creepymonster/GlucoseDirect/blob/31e1e902a5e66fbf6d0f4b8eb03f88624f322572/App/Modules/SensorConnector/LibreConnection/LibreConnection.swift#L49-L51 is executed and the sensor is reported wrongly as in “starting” phase, like some users are reporting on Facebook on troublesome scans.

By ascertaining that the age is not zeroed
```
} else if result.sensor.age > 0 && result.sensor.age <= result.sensor.warmupTime {
                    sendUpdate(age: result.sensor.age, state: .starting)
}
```
the expired Libre 3 state is detected correctly:

[image]

creepymonster, 2023-01-14:
Hello gui-dos,

thank you for your message - I'm always happy to hear from you.

I suspect the problem is more difficult to solve. Right now I can't get the age of the sensor from the NFC scan of the Libre 3. I don't get the real age of the sensor until I retrieve the data from LibreLinkUp.

Therefore, I set the age of the sensor to 0 during the NFC scan, which means that the sensor is set to "Starting" until the first data retrieval. I do this because I only allow connections to the sensor when the status is "Starting" or "Ready".

https://github.com/creepymonster/GlucoseDirect/blob/31e1e902a5e66fbf6d0f4b8eb03f88624f322572/Library/Content/Sensor.swift#L218

If I would not do this, the sensor might not be connected otherwise, and then the real age of LibreLinkUp could not be retrieved.

It would be better to determine the age from the NFC scan. Maybe you have a tip.

Many greetings
Reimar

OddSquirrel, 2023-01-14:
Thanks for the explanation, that "starting" message from an already started sensor always puzzled me. 😋

gui-dos, 2023-01-14:
This was exactly my first intent when I started to scrape LibreLinkUp since the `activationTime` and the `receiverId` are to be passed also to the NFC `A8` command to activate a sensor. You compute the `receiverId` starting from the `patientId` which can be also captured through a web proxy like Charles, but I like to eat my own dog food :-)

If you scan a baby sensor in its early minutes of life you can notice that after a dozen minutes its state switches immediately from `02` (`PATCH_STATE_INSERTION_DETECTION`) to `04` (`PATCH_STATE_PAIRED`)  and this is the new operational state for its next 14 days.

`03` is now an error code which is defined as `PATCH_STATE_INSERTION_FAILED` and this is why I am decreasing by one the next state codes to convert them to the current definitions.

Notice however that they are defining a third slightly different state enum  (`1: MISSING, 2: WARMUP, 3: READY, 4: EXPIRED, 5: ACTIVE, 6: ENDED, 7: INSERTION_FAILED`) which is sent via BLE as a field through the new “patch status” characteristic. Unfortunately it seems that our code cannot be more abstract and less messy than theirs… :-)

As a first step I think you should define such two new `State` enums specifically for a new Libre 3 class. They are not obfuscating their definitions in the Android APK and you know you can always count on my well tested conversions to iOS :-)

gui-dos, 2023-02-11:
@creepymonster

I discovered that the custom NFC command `A0` returns the activation date of an already activated sensor.

Here you can find a first implementation in https://github.com/gui-dos/DiaBLE/blob/23d23cb7647ecc38a89edb3de3991c36404d39e5/DiaBLE/Libre3.swift#L666-L677 and how to parse its output in https://github.com/gui-dos/DiaBLE/blob/23d23cb7647ecc38a89edb3de3991c36404d39e5/DiaBLE/NFCTools.swift#L296-L331

Notice that this way you can get also the BLE PIN which is involved when building the responses to the security challenges!

You have to pay attention when playing with its cousin command `A8` because it changes the BLE PIN of the sensor and you have to reinstall Trident to be able to reconnect to it.