creepymonster/GlucoseDirect #518: Cannot connect to sensor

Bootsy3, 2023-02-16:
Hi
I have Libre 3. I can see my data in the Libre 3 app and the LibreLink App. I have put my librelinkup password etc in the glucose direct app. I have disconnected Bluetooth to Libre 3 but I keep getting error saying missing user or token

yesterday it said “pairing” and nothing happened. Today it just says disconnected and when I scan the sensor I just get error message 

what am I doing wrong?

iDustbin, 2023-02-16:
Unfortunately I’m running into the same issue :( any hint would be helpful!

Bootsy3, 2023-02-16:
I logged out of my Libre linkup and changed my password.  Then logged back on with new password. Put this password into GD. Made sure Libre app was connected to Bluetooth and librelinkup was connected to the Libre app. I could see my readings on both devices. Then tried scanning glucose direct and this time it worked. Waahoooo.

Bootsy3, 2023-02-16:
Issue resolved for me