creepymonster/GlucoseDirect #532: Missing User or Token

JulesJulws, 2023-04-07:
Hi,

many thanks for this great invention. But for a few weeks now, I am not able to connect my sensor. It always shows the error message missing user or token.
Libre 3 and Libre Link works properly and the correct Libre Link credentials are entered in GD. 
Could you please help? I really want to use GD again!
Many thanks!

daringlyorganic, 2023-04-09:
Same issue for me also.

DAcodedBEAT, 2023-07-18:
Is this still an issue?  Building from source seemed to work for me

JulesJulws, 2023-07-19:
Hi, yes it is. I deleted everything and started ever  again. Now I receive the following error message: Von meinem iPhone gesendet

DAcodedBEAT, 2023-07-19:
@JulesJulws I don't think your error message was set correctly.  Can you send your message on Github?

JulesJulws, 2023-07-28:
It seems I need an invitation? 

[image]

DAcodedBEAT, 2023-07-28:
@JulesJulws The version on TestFlight is outdated so even if a TestFlight invite was open, the code would be outdated and would not be able to get the data from Libre Link Up (which has your Libre 3 data).

In order to get this app working, I had to update the code to make the application buildable in Xcode (https://github.com/creepymonster/GlucoseDirect/pull/564), build the code manually on my machine, and scan the sensor to login.

If @creepymonster can make a new release and push it to TestFlight, you can try to use that (assuming TestFlight invites become available) but he seems to be inactive on GitHub.

zestysoft, 2023-10-07:
@DAcodedBEAT I'm running into this issue too.  It happened after swapping out my sensor for a new one.  I was on holiday and couldn't swap out the sensor for two+ days if that matters? Everything was working before I left -- I had recompiled and installed the app after updating to ios 17.

Just today, I deleted the app from the phone, did a `git pull` and re-built the app, and re-inputted the librelink creds (I use a password manager for this, so i know that hasn't changed) but after scanning the sensor, still receive the same "Missing user or token" error.

Previous to this, I went into "Manage Connections" in LibreLinkUp, removed myself, and then re-added in case there was some kind of stale token thing going on.

The Libre 3 app and LibreLinkUp apps show current data from the sensor.

Details:
iPhone 14 Pro Max running ios 17.0.3
LibreLinkUp 4.8.0.10

zestysoft, 2023-10-07:
Okay, I set a breakpoint to look at the results from logging in:
[image]

Looks like it wants to redirect somewhere?

gui-dos, 2023-10-07:
The response in JSON format in such a case of redirection a very simple, for example:

```
{"status":0,"data":{"redirect":true,"region":"fr"}}
```

(https://github.com/gui-dos/DiaBLE/blob/4cf7e0ec4cfa6017aa60fe476d068b833e003909/DiaBLE/LibreLink.swift#L148)

zestysoft, 2023-10-07:
Okay, I've stepped through the code a bit -- here's an example of the tou request response decode. I've truncated the token:
```
[GlucoseDirectApp/LibreLinkUpConnection.swift:310][tou(apiRegion:authToken:)][isMT: false] LibreLinkUp tou, response: Optional("{\"status\":4,\"data\":{\"step\":{\"type\":\"pp\",\"componentName\":\"AcceptDocument\",\"props\":{\"reaccept\":true,\"titleKey\":\"Common.privacyPolicy\",\"type\":\"pp\"}},\"user\":{\"accountType\":\"pat\",\"country\":\"US\",\"uiLanguage\":\"en-US\"},\"authTicket\":{\"token\":\"eyJhbGciO\",\"expires\":1696702627,\"duration\":3600000}}}\n")
```

So I guess it's not returning the user?

gui-dos, 2023-10-07:
The usual trick is not a development matter: just logout and login again to your LibreLinkUp account and accept the new "Terms and Conditions".

gui-dos, 2023-10-07:
I wrote a Swift Playground for the iPad and Mac still available at https://github.com/gui-dos/DiaBLE-Legacy/tree/main/DiaBLE%20Playground.swiftpm to test the error messages without necessarily using Xcode.

Recently when I upgraded to LibreLinkUp 4..8.0 I received also this error message 

[image]

(Yep: I am connecting to the lib eview.io servers also directs from my AppleWatch B-) )

Luckily the day after the connection was working again but it was their side's fault..

DAcodedBEAT, 2023-10-14:
> Okay, I've stepped through the code a bit -- here's an example of the tou request response decode. I've truncated the token: `[GlucoseDirectApp/LibreLinkUpConnection.swift:310][tou(apiRegion:authToken:)][isMT: false] LibreLinkUp tou, response: Optional("{\"status\":4,\"data\":{\"step\":{\"type\":\"pp\",\"componentName\":\"AcceptDocument\",\"props\":{\"reaccept\":true,\"titleKey\":\"Common.privacyPolicy\",\"type\":\"pp\"}},\"user\":{\"accountType\":\"pat\",\"country\":\"US\",\"uiLanguage\":\"en-US\"},\"authTicket\":{\"token\":\"eyJhbGciO\",\"expires\":1696702627,\"duration\":3600000}}}\n")`
> 
> So I guess it's not returning the user object in the way GD expects?
> 
> Ahh, here's where things break I think since it's not returning an id: [guard let userID = loginResponse.data?.user?.id](https://github.com/creepymonster/GlucoseDirect/blob/816f993f6872a6058a716597c7b224499db259fc/App/Modules/SensorConnector/LibreConnection/LibreLinkUpConnection.swift#L212)

Did get this sorted out? My confusion is that `"status":4` is believed to be the state when the tou isn't accepted and you seem to be getting this response after attempting to accept the tou

DAcodedBEAT, 2023-10-14:
> Okay, I've stepped through the code a bit -- here's an example of the tou request response decode. I've truncated the token: [...]

@zestysoft Did get this sorted out? My confusion is that `"status":4` is believed to be the state when the tou isn't accepted and you seem to be getting this response after attempting to accept the tou

zestysoft, 2023-10-15:
@DAcodedBEAT Unfortunately no.  From what I can see, it connects to the authentication server, gets a redirect, connects to that, gets a 200 with some details, and then reconnects a third time using those details which also returns a 200 with an object like you see above.

If I enter in the wrong password, I get an entirely different result, so I'm pretty certain my credentials are correct, or correct enough to auth into librelinkup.

I just checked my nightscout instance and it looks like librelinkup is returning similar status 4 info to that too. I'm going to delete that librelinkup account through their app and just re-do this whole thing.

zestysoft, 2023-10-15:
Okay, deleted the account through their app (again, I was receiving up to date info in that app).  I had to enter in the password to nuke the account so I know the creds were valid.

After recreating, everything is working again.  I wonder if I should have logged out in the librelinkup app and re-authed -- I've ran into similar token issues that required using a web browser to log in and accept updated EULAs -- maybe it would have prompted me for something like that.