creepymonster/GlucoseDirect #539: TestFlight Beta

AuspiciousLynx, 2023-05-20:
Hi, (English Below)
Ich gehe mal davon aus das du deutsch sprichst wenn du aus Berlin kommst, falls nicht schreibe ich das darunter nochmal auf Englisch.
Ich schaue nach einem Platz in der Testflight Beta für meine Oma mit Diabetes. Sie hat Version 3 und Probleme mit der Verbindung bei der Offiziellen App. Sie nutzt auch eine AppleWatch, das es keine offizielle App gibt findet sie noch immer schade.
Wäre klasse wenn es irgendwie möglich wäre die App doch zu nutzen. Danke

I am asking for my Grandma, she is using the EU freestyle Libre 3 system but is having problems with the connection to her phone so I wondered if your app would be better in this. She is also using an Apple Watch, she could be a good tester and even help you giving problems to you. Is there any way to find one last place in the TestFlight app?

thank you

mayorgergich, 2023-05-21:
I didn't want to create a new issue with the same request, so OP I hope you don't mind I comment on yours. 

My husband has been using the sensors for quite a while and has been using new meds which help him with his diabetes but he's had the hardest time regulating dosage. If there's another spot in the beta it would help us so much in better understanding how his medication is affecting his glucose levels.

DAcodedBEAT, 2023-07-28:
I'm not a maintainer of the project so I do not have any TestFlight invites but I can speak to the capabilities of the application from looking at the source code from this repository, which might help clear up any confusion about the application.

@AuspiciousLynx Here are the currently supported Libre sensors (from the README):
>* Libre 3 Support (over LibreLinkUp)
>* Libre 2 EU support (without transmitter) - Currently not supported are Libre 2 US/CA/AUS, Libre Pro, Libre H sensors :'( - I am very happy about support for their connection
>* Libre 1 support with transmitter (supported transmitter bubble)
>* Libre 1, manual scans without transmitter

Since your Grandma is using the EU freestyle Libre 3 system, she will need to still use the Libre 3 app to upload the data to LibreLinkUp, since GlucoseDirect can only read the data from Libre 3 via LibreLinkUp's service (not directly from the sensor).