creepymonster/GlucoseDirect #552: Values not retreived anymore after LibreLinkUp App update to version 4.7

J0ANMM, 2023-06-17:
I was forced to update yesterday the LibreLinkUp App to version 4.7.

Since then GlucoseDirect is not able to pull and display any data anymore.

[image]

I am using an iPhone XS with iOS Version 16.5.

Should I do something to fix the error after the update of LibreLinkUp?

OddSquirrel, 2023-06-17:
Same here. They just can't leave well enough alone over at Abbott. 😤🧨💣

Restarts didn't do anything, and there weren't any new terms to accept, either. Everything ends up in Libre LinkUp, but GlucoseDirect can't grab it for some reason.

iPhone 14 Pro Max running iOS 16.5.

FokkeZB, 2023-06-26:
Did you try a fresh build of the current main branch? It received multiple LibreView related updates recently, including for 4.7 and automatically accepting TOU.

mpvbasilio, 2023-06-27:
How do we get a fresh build? Sorry totally a newbie here. Just a simple user to get my data over to apple health.

J0ANMM, 2023-06-27:
It is now solved for me.
When I entered into the GlucoseDirect app (installed via TestFlight), an update notification was shown and everything works fine again.

(This update was not shown when I opened this issue last week, but at some point after that.)

miguelruivo, 2023-07-13:
Hi, can someone tell me where is the option to add the Libre LinkUp connection in the Glucose Direct app?

FokkeZB, 2023-07-13:
@miguelruivo AFAIK you need to scan a FSL 3 sensor to unlock it. It would be nice to make it an option for any sensor if somehow you prefer going via Libre LinkUp/API.

miguelpruivo, 2023-07-17:
For me that would be perfect @FokkeZB. I plan to use the GlucoseDirect as a "bridge" only between Libre LinkUp and calendar events, when the new update kicks in that will turn the Libre app into CGM.

gui-dos, 2023-07-17:
> @miguelruivo AFAIK you need to scan a FSL 3 sensor to unlock it.

No: Glucose Direct doesn’t activate a Libre 3 whey scanning the sensor, it is just getting its serial number and state from its new extended `patchInfo` (24 bytes instead of 6).

I discovered that when you send the NFC custom command `A0` (I still have to understand which `A0` and `A8` are `CMD_ACTIVATE_SENSOR` and `CMD_SWITCH_RECEIVER`: https://github.com/gui-dos/DiaBLE/blob/58fc7b288faa392ac4448dc04aa4c645c9712744/DiaBLE/Libre3.swift#L765) you can get already the activation time avoid a first trip to the LibreLinkUp servers. And the BLE PIN also returned is a great bonus! ;-)