creepymonster/GlucoseDirect #563: Build Failed - Error: Value of type 'EKEventStore' has no member 'requestFullAccessToEvents

grostecki, 2023-06-28:
When building in xcode I get the error "Value of type 'EKEventStore' has no member 'requestFullAccessToEvents"
[image]

w-aurelien, 2023-07-05:
Exactly the same. Any solutions ?

w-aurelien, 2023-07-05:
@grostecki Solved using xcode 15 beta

grostecki, 2023-07-05:
I have Monterey 12.6.6 and Xcode 14.2. I can't update anymore. Unfortunately the error is still there.

grostecki, 2023-07-09:
Ok. I got it.
MacOS Monterey 12.x + XCode 14.3 - Build Failed
MacOS Ventura 13.4 + XCode 14.3 - Build Failed
MacOS Ventura 13.4 + XCode 15 beta3 - Build Success

DAcodedBEAT, 2023-07-28:
@grostecki if possible, can you re-open this issue?

I'd really like some visibility on this until either the fix from https://github.com/creepymonster/GlucoseDirect/pull/564 gets merged or Xcode 15 goes out of beta.

grostecki, 2023-07-29:
[image]

grostecki, 2023-07-29:
Your fix
[image]

grostecki, 2023-07-29:
Build success
[image]

gui-dos, 2023-07-29:
These days unfortunately the latest Xcode 15 beta 5 doesn’t allow publishing on TestFlight: “Xcode 15 Beta 5: Unsupported Xcode or SDK Version” (https://developer.apple.com/forums/thread/734618).

Hopefully this is the reason why @creepymonster  is not committing this PR yet. We share most of the code base: allow me to spam my project https://github.com/gui-dos/DiaBLE/tree/main/DiaBLE%20Playground.swiftpm: the Playground version runs directly on a Mac or iPad and should let you test the BLE and LibreLinkUp connections showind details log on the central Console Tab in the meantime B-)