creepymonster/GlucoseDirect #568: Add libre linkup connection?

miguelpruivo, 2023-07-13:
How do I add libre linkup connection to the app? I can’t find any option to add that connection only to nightscout. 

Thanks!

Roertoo96, 2023-07-21:
I have the same problem. After my first deployment, I noticed the linkup connection option. However, after testing some settings, it disappeared and hasn't returned. I also tried redeploying the app, but it didn't fix the issue.

Roertoo96, 2023-07-21:
Ich habe die Lösung gefunden. Du musst in den Einstellungen oben auf ohne Transmitter stellen. Dann einmal versuchen deinen Sensor zu Scannen. Danach kommt eine Meldung das die Credentials fehlen und de Felder erscheinen in den einstellungen

J0ANMM, 2023-07-29:
The same just happened to me: after playing around in the settings and changing from “without transmitter” to “bubble transmitter”, the fields to introduce my LibreLinkUp details just disappeared.

Now, even after switching back to “without transmitter”, I do not see the fields for the LibreLinkUp anymore and in the overview the status is “Disconnected”.

(Note that I started to play around with the setting after getting a new sensor started but still having as a status “sensor ended” in GlucoseDirect)

BenelliFurtado, 2023-07-30:
Same here, I'm using 1.0.0. No missing credentials message after scanning, just an error that pairing did not work (Libre 3).

Roertoo96, 2023-07-31:
No, after I tried to scan my libre 3, the login field was in the Settings section. Try switching the transmitter a few times and scanning again. I played around a lot until it showed up.

J0ANMM, 2023-07-31:
When I opened the app today, the option “Scan sensor” showed app in the overview. (Maybe it is shown after X hours with status “disconnected“?)

In any case, after scanning the sensor the issue has been fixed.

maddox, 2024-04-10:
For me, the only way I can get data is if I scan the sensor, which doesn't really make the libre linkup support make much sense. 

What's worse is, whenever my sensor expires, I have no way to add another one. The only way I can get it to work is by deleting the app, and downloading it again from TestFlight, then going through everything all over again including setting all the settings. 

It's quite cumbersome to do every 2 weeks.