creepymonster/GlucoseDirect #570: Apple Watch

robertherber, 2023-07-15:
Would love to use this standalone on my Apple Watch (using Libre 2). Is it possible?

DAcodedBEAT, 2023-07-15:
As of now, if you want to have current glucose levels shown on the watch, you can Export to Apple Calendar and add a calendar watch complication to see the level, as described in https://github.com/creepymonster/GlucoseDirect/discussions/466

It looks like there was some development effort to have a native watch complication from the app directly (https://github.com/creepymonster/GlucoseDirect/pull/506 and https://github.com/creepymonster/GlucoseDirect/pull/508) but it hasn't landed in a development build yet.

robertherber, 2023-07-16:
Hi @DAcodedBEAT,

To elaborate: What I'm digging for here is a way to leave the phone at home and just use the Apple Watch by itself. This would require an Apple Watch app that has a direct bluetooth connection to the sensor.

I'm guessing the bluetooth connection implementation could be 99% reused on the watch. Although the initial step (scanning to connect to the sensor) might still have to be done from the phone?

I think using the Calendar might very well be the best way to display it still. If things haven't changed since I last looked at complications are a bit rate limited to preserve battery, while the calendar always updates instantaneously.

gui-dos, 2023-07-17:
Yes, it is possible to reconnect to an European Libre 2 from a different device than the iPhone it was initially scanned on: an Apple Watch, a Mac, an iPad… and I think even an Apple TV or a Vision Pro since they implement the same Core Bluetooth API (I didn’t test yet… B-).

There are some parameters that have to be encoded and sent via Bluetooth to reconnect to a sensor, in particular a progressive counter that has to be incremented by one at each new connection: this is the simple reason why you loose the alarms on other apps when you scan the Libre 2 to activate the BLE streaming of data since such counter is reset to 0 and the sensor stores it internally.

Luckily you can scan a sensor via NFC to activate the BLE streaming of data countless times and it has been discovered that any Int32 can be used as the “unlock code”: Abbott uses the timestamp of the activation time but you can see that all the current iOS (and Android) apps have chosen `42` as the ultimate solution for anyone ;-)

In my open source project  https://github.com/gui-dos/DiaBLE I let the user input these parameters manually (along the ones to calibrate the sensor established on factory and scannable via NFC) since the Apple Watch version is really independent and doesn’t need a connection to an iPhone or iCloud at all B-)