creepymonster/GlucoseDirect #578: Not getting High/low alarm notification on watch.

3djc, 2023-09-02:
My setup:
- Libre 2 sensors (sensor was started using the standalone Freestyle 2 reader)
- No Libre app on phone
- iPhone XR and Watch 8, all latest versions

What works
- everything works fine on phone
- glucose level display on watch using calendar
[image]
- when phone looses contact with sensor, an alert notification is sent to watch, and **this alert works fine** (watch vibrate and all)

What does not
- when either low or high limits are crossed, the phone vibrate (or rings depending on settings)
**- BUT i receive no notification on watch**

App is setup like this
[image]

[image]

3djc, 2023-09-04:
Is there a way to access the logs ? Unfortunately, I don't have a Mac, but looking at the code, I wonder if isSnooze is not wrong for me likely because I did something wrong somewhere

jkyyhkynen, 2023-09-07:
I have the same issue. Notification mirror from Glucose Direct is enabled, but not single notification has worked on the watch. On phone all notifications work.

3djc, 2023-09-08:
> I have the same issue. Notification mirror from Glucose Direct is enabled, but not single notification has worked on the watch. On phone all notifications work.

Do they ? Because I noticed I get the audio alarm on phone, but actually no notification. So to me, the issue is not that notification aren't sent to watch, they are not happening on phone either (but alarm do)