creepymonster/GlucoseDirect #581: TestFlight beta is full

tris, 2023-09-08:
Uh oh -- using the join link in the README leads to a page that says "This beta is full".

Any workaround for that, aside from manually compiling the app?

3djc, 2023-09-10:
Slots are becoming free, just at a slow pace. Keep trying TestFlight, or look for an announcement of free slots in th FB group. Took me less than a week to find one

mxsrm, 2023-09-30:
Beta is still full. :/

CharachorderTutor, 2023-10-01:
What would it take to build a GlucoseDirect2 Testflight? Is it just the $100 developer fee? How do I go about doing this if that is the only limitation?

wrtsprt, 2023-11-08:
I guess you could clone the project, become an Apple Developer and setup your own TestFlight. 

I am wondering if another avenue might be more interesting though: if the app would be buildable yourself without the need for NFC / Real Time Notifications, ppl could build it themselves without needing a developer account. IMO this should be pretty poplar. I'll raise an issue around this.

AvasDream, 2024-01-08:
Hi @creepymonster, I have an Apple Developer Account and should be able to set up the CI/CD (in my cloned repo) to publish a new version to Test flight. With this, another, 10000 People could be added to the Beta and I could provide you with access to my Test Flight Project. 

Would this be ok for you?