creepymonster/GlucoseDirect #582: Incompetent

dugamarian, 2023-09-11:
I don't know who currently manages the app and Facebook group, but this Marcel seems to be a baker, not a developer. He doesn't seem to know what world he's living in.

jordanlambrecht, 2023-09-25:
@dugamarian Okay, well, that's a pretty rude thing to say. Any specific examples of them being incompetent, or are you just being a d*ck to be a d*ck? They're volunteering their time and doing it for free simply because they're passionate about the project. 

Also, developers notoriously make TERRIBLE community/project managers.

Finally, you have a total of 3 github contributions in 2023. It's not like you're doing your part.

dugamarian, 2023-09-26:
That message was NOT addressed to you or to Mr. Reimar. It is addressed to Mr. Marcel S, who I don't know how he ended up managing that application. I'm sorry that such people, who have no clue, end up giving people incorrect advice.
I repeat, that message was for Mr. Marcel S, and it was not meant to offend any other contributors to the application.

jordanlambrecht, 2023-09-27:
@dugamarian I'm aware of who it was directed towards. It's still rude. You have also yet to clarify what they have done to make you feel that way.

dugamarian, 2023-09-30:
I have been removed, arbitrarily, from the testing group twice, even though I actively test the application 24/7, but I didn't say anything, maybe I hadn't donated enough. What frustrated me is that I provided feedback on the FL3 sensor calibration in the Facebook group, and as a result of that, I was banned from the group without the right to dialogue, simply and once again, arbitrarily.

In conclusion, perhaps I was indeed impolite, but I also had reasons.

closed 2023-12-15 by dugamarian