creepymonster/GlucoseDirect #605: no connection settings available in latest update

greghao, 2024-09-19:
In the [Settings](https://github.com/creepymonster/GlucoseDirect/raw/main/Screenshots/settings.png?raw=true) photo, it shows a "Connections Setting" where users can input their LibreLinkUp email/pass but on the latest version downloaded from TestFlight, 1.0.0 (187) this section appears to be missing now. Is there something I'm missing? I've deleted and downloaded from TestFlight twice already.

Errator, 2024-12-19:
Hi, i have the same Problem. Could you fix it? I have 1.0.0 (191)

greghao, 2025-01-20:
@Errator I changed to a libre 3, and upon scanning the cgm, the login/pass for librelinkup showed up in settings. it looks like it is contextually aware. Hope that helps.

closed 2025-01-20 by greghao