creepymonster/GlucoseDirect #609: Adding Libre 3 sensor

christoffer22, 2024-10-03:
Hi,
I am all new to GlucoseDirect. I am currently wearing a Libre 3 sensor, with 11 days remaining. Is it possible to add this sensor to the GlucoseDirect app? Or do I have to replace the sensor with a new one, and start it / scan it for the first time via the GlucoseDirect app, and not the Libre app?

I have scanned the sensor, and I have entered my Libre Link Up details. But it just says "No Data" and state is "Scanning". It also says that the sensor is in starting phase, with 1h warm up. But this is not correct because I am already using the sensor, with 11 days remaining?

[image]

CallumMcK, 2024-10-03:
@christoffer22 This can occasionally happen on first scan. If you close the app and open it again does it still say No Data? 

If it does, try force shutting the app (slide up) and reopen - try rescanning then.

gui-dos, 2024-10-04:
The culprits are the lines
https://github.com/creepymonster/GlucoseDirect/blob/e3636d0e47cd490bb02db9e10b984bc0b53ca70d/App/Modules/SensorConnector/LibreConnection/LibreLinkUpConnection.swift#L264
and
https://github.com/creepymonster/GlucoseDirect/blob/e3636d0e47cd490bb02db9e10b984bc0b53ca70d/App/Modules/SensorConnector/LibreConnection/LibreConnection.swift#L49
because they don't ascertain that `sensorAge` and  `result.sensor.age > 0`.

I reported this bug multiple times but unfortunately @creepymonster didn't reply to me anymore even privately... ;-/

maddox, 2024-10-14:
I’m having the same issue after things working for almost two years. It seems it might have started with iOS 18. 

Force quitting the app doesn’t do anything. 

It doesn’t just say No Data, it also says Missing User or Token.

maddox, 2024-11-23:
Is this something anyone else is experiencing? I haven't had this app work for months.