creepymonster/GlucoseDirect #611: Unknown error

ktrshi, 2024-11-15:
Hello, today I encountered an issue in the app where blood sugar data is not being transferred. The data is available in LibreApp and LibreLinkUp, and I haven’t changed any settings. Here’s a screenshot.
[image]

mrabus, 2024-11-15:
I face the same behavior since about two hours. Have scanned the sensor multiple times, hard reset the app and even restarted my phone. 
Nothing lead to any improvement.

ktrshi, 2024-11-15:
I've seen several notifications from LibreLinkUp about the new update. Perhaps the error occurred because of this.

mrabus, 2024-11-15:
In the facebook group (glucose direct) the admin posted that LibreLinkup doesn't support following anymore. That is the cause of the issues.

ktrshi, 2024-11-15:
It's very sad to hear that. It looks like it's over for third-party projects

mrabus, 2024-11-15:
Same for me. App works after updating it from TestFlight. Brilliant!

sn0wh4ze, 2024-11-16:
Update GlukoseDirect über Testlight dann funktioniert es.